opulo-inc/feeder #5: Route RS485 as differential pair

G-Pereira, 2023-03-16:
Improve RS485 layout.
[Here](https://www.ti.com/lit/an/slla272d/slla272d.pdf) is a Texas Instruments app note as reference.

sphawes, 2023-04-19:
Hey @G-Pereira! Are you referring to trace length matching? RS485 is a differential signal, but it doesn't seem to require being routed with trace length matching. We're only driving the bus at 57600 baud at a distance of ~2m, so we're well under seeing the transmission line effect. It couldn't hurt to do, but I don't think it's necessary.

G-Pereira, 2023-04-21:
Not specifically, I mean routing them side by side with KiCAD's bult-in feature for routing differential pairs. It's one of those things that takes one more minute to do it right and a pain to change down the road when you start having issues.
This is a good article if you want to learn more: https://resources.altium.com/p/what-are-differential-pairs-and-differential-signals

sphawes, 2023-05-09:
Yup, I understand the routing feature that you're referencing, it just isn't necessary for the speeds that we're running RS-485 at. That being said, I can't see any harm in routing them like that!