opulo-inc/feeder #6: Feeder Floor gerbers are only for board 50, not the full panel

qwertymodo, 2023-04-07:
The provided gerber files for the feeder floors only contain the design for a single board.  The full panel design doesn't seem to be available.

[image]

sphawes, 2023-04-19:
hey @qwertymodo! the feeder floor pcb is electrically the same for all 50 (aside from the last one which populates a 120R resistor in R1). I left the 50 on there as more of a note that 50 are required. On next uprev i can change it to a white block of silkscreen so it provides a nice place to write it in, and so it's less confusing

willyvmm, 2023-05-27:
If someone wants the gerbers ready for production: [gerbers.zip](https://github.com/opulo-inc/feeder/files/11581843/gerbers.zip)
(not tested yet!)

lucian151, 2025-01-10:
Fixed by moving the design to blades in the LumenPnP repo