opulo-inc/feeder #11: No inrush current protection

sphawes, 2023-05-09:
# Version Number
v1.0.1

# Bugfix or Enhancement
Enhancement

# Description
Folks in Discord recommended that we add a form of protection against high inrush current on the feeder controller. 

# Suggested Solution
The current thought is to add a ferrite bead inline with the 24v rail directly after the spring pin interface will help reduce the current spike when a feeder is loaded onto a machine.

Would love any thoughts about how best to implement this!

[image]
[image]

Magpie-81, 2023-05-10:
A ferrite bead is definitely a good idea, but this mostly helps to filter out high frequency noise on the supply line.
An inductor would help better to limit inrush currents, and if tuned properly with some decoupling capacitors helps even more to filter out low frequency noise, as a ferrite bead only builds up its impedance at pretty high frequency (impedance values for ferrites are specified at 100MHz!). There are specific ferrites that have a better performance (less impedance drop-off) at lower frequency, but you would have to very specifically select this single source component.

A proper hot-plug and inrush limited design would need some semiconductor in line. You can build that with a mosfet and a few resistors/capacitors, but getting a slew rate limited load switch is usually easier to use and does not need more board space. Something like the TPS1641 would be a possible candidate for this application. 

Downside is that this would push the BOM cost up, as you probably need to invest 1-2$ for either a high inductance value coil or the active IC solution.

warasilapm, 2023-05-10:
@Magpie-81's thoughts are similar to mine. FB is a good idea for EMI, but it doesn't provide a true solution. A choke inductor would probably be better here, and a large enough value would help with inrush as well. Since the currents are the cost doesn't have to be anything fancy. Something as simple as [IHLP-4040DZ-1A](https://www.digikey.com/en/products/result?s=N4IgjCBcoLQCxVAYygMwIYBsDOBTANCAG4B2aWehA9lANohwCcA7AKwBMIAuoQA4AuUECAC%2BYoA) would do the trick here, I think.

sphawes, 2023-05-10:
Awesome, thanks for the thoughts! I like the idea of just using an inductor and not needing to add another IC to the BOM.

I'm curious what y'all think of this approach:
[image]
It's very inexpensive, limits charging the cap which is the thing causing the initial current spike, and shouldn't affect the load's ability to pull from the cap because of the diode.

Perhaps this in conjunction with a FB or inductor would be ideal?

warasilapm, 2023-05-10:
Hi I just wanted to document that I have inrush issues at around 13 or 14 feeders on my v2 kit build.

Maybe I'm a good guinea pig then?

sphawes, 2023-05-11:
@warasilapm just to clarify from our discussion earlier in discord, you only have an issue when powering on the machine with 13 or more feeders _already mounted_ beforehand? If you power on the machine with _no feeders mounted_ and insert them one at a time after the Lumen has booted, everything works as intended?

warasilapm, 2023-05-11:
If 13 or more feeders are mounted before powering on, the machine fails to power on.

Magpie-81, 2023-05-11:
> Awesome, thanks for the thoughts! I like the idea of just using an inductor and not needing to add another IC to the BOM.
>
> I'm curious what y'all think of this approach: [image] It's very [...]

The issue with using the diode/resistor solution would be that you limit the function of the capacitor significantly. There are no ideal diodes, they all have a forward voltage drop, and since we expect our load to be using pulsed currents and not just a constant current flow, you risk to produce quite the voltage ripple with this solution. You could probably try to add more filtering between the bulk capacitor and the load to remove most of that ripple again, but this seems to just blow up the BOM even more.
Seems to be kind of a "hacky" solution and would have to tested and verifyed properly before you fully commit to this and sell boxloads of feeders with that hack on board.

FailSafinho, 2023-05-11:
What about using NTC as the inrush current limiters? When they are cold they will have high resistance, so when you power the circuit up, it will limit the amount of current going through the caps, then, when it's warm, the resistance will drop.

warasilapm, 2023-05-11:
> > Awesome, thanks for the thoughts! I like the idea of just using an inductor and not needing to add another IC to the BOM.
> > I'm curious what y'all think of this approach: [image] It's very inexpensive, limits charging the cap which is the thing causing the initial current spike, and shouldn't affect the load's ability to pull from the cap because of the diode.
> > Perhaps this in conjunction with a FB or inductor would be ideal?
> 
> The issue with using the diode/resistor solution would be that you limit the function of the capacitor significantly. There are no ideal diodes, they all have a forward voltage drop, and since we expect our load to be using pulsed currents and not just a constant current flow, you risk to produce quite the voltage ripple with this solution. You could probably try to add more filtering between the bulk capacitor and the load to remove most of that ripple again, but this seems to just blow up the BOM even more. Seems to be kind of a "hacky" solution and would have to get tested and verifyed properly before you fully commit to this and sell boxloads of feeders with that hack on board.

I couldn't quite put my finger on what was bad about this yesterday when I saw this. Glad @Magpie-81 had their finger on the pulse... or ripple in this case!

If you're willing to add a dollar or so to the BOM, an inductor or an IC similar to what was suggested earlier (though perhaps one more widely distributed) is the way to go. Buy once, cry once, IMO.

sphawes, 2023-05-12:
Spent most of the day yesterday looking at different IC solutions, but most were prohibitively expensive or took up too much board real estate to fit. Using an inductor solves both of those problems.

I've added a Sunlord SWPA5020S470MT 47uH inductor inline with the 24v rail on the feeder:
[image]

According to the [datasheet](https://datasheet.lcsc.com/lcsc/2110091630_Sunlord-SWPA6045S470MT_C36414.pdf
) it can handle up to 770mA of current steady-state, which should be plenty for each feeder. [Here's a rough simulation](https://tinyurl.com/2hyagma6) to show the difference between the current circuit and one with the inductor inline. There's certainly an appreciable difference in current spike, but the resonance is a bit concerning to me. What do y'all think about this approach?

warasilapm, 2023-05-12:
Ultimately, I don't think an inductor is going to be sufficient inrush protection. With this much bulk capacitance, the charge demand on the power supply for multiple feeders is going to be immense. I was talking with my coworkers at work and we had the following thoughts:
1. There is too much bulk capacitance on the 24 V side of the buck regulator. You may be able to go as low as 10 uF. The buck regulator you chose only shows 100 uF in the data sheet, but it may be possible to use less. This also opens up better performing X5R ceramics which also take up significantly less space. This is the only affordable way to reduce the start up charge demand. The buck regulator doesn't have a soft start to prevent the following 470 uF from causing a problem but it is current limited so this should help.
2. Add a smaller (2.2 uH) series inductor and capacitor (1 uF) to the input of the device in a C-L-C pi filter arrangement. This limits the inrush current somewhat but more importantly provides a substantial (~ 20 dB) reduction in conducted emissions and immunity.

Here's the simulation I've been playing with. It's got a bunch of parasitics added and also a very crude noise model of the switching regulator load to get an idea of the noise involved.

30350n, 2023-05-13:
The 470uF should probably also be replaced with a 100uF, just because of availability like described in #1 . The 100uF should work completely fine, but will result in a slightly noisier output voltage, which shouldn't be a problem though.

sphawes, 2023-05-17:
> I think it would make more sense to do the majority of the inrush limiting on the mobo since it has about 800 uF of bulk capacitance even without feeders.

I agree with this. Ultimately, the surefire solution is to put a dedicated e-Fuse IC on the motherboard that limits the max current of the 24v rail. As @warasilapm said, we already have a ton of bulk capacitance on the motherboard that we should also be addressing. We also don't have a ton of space on the feeder board left, and it makes quite a bit more sense to have the inrush limiting at the source instead of up to 50 times across the rail. That being said, we should still add features to prevent this on the feeder as well.

In regards to reducing the capacitance on either side of the buck converter, I could see trying to reduce them, but I'm hesitant to stray from the suggested values in the datasheet. As @30350n said, the 10v rail doesn't have to be particularly noise-free, so I could see dropping the values a little (especially on the 10v side). I'll run some tests with some different values and see how significantly it affects the integrity of the rails.

Here's my proposal: We have a rather large inductor in series as I showed above, ideally with the Pi filter that @warasilapm suggested as well (thank you for the simulation!). I'll also work to incorporate a solution similar to @TPwire's suggestion in the Lumen motherboard, and source a power supply that does a better job handling inrush current. I've been testing some better quality ones and they actually eliminate the boot issue altogether.

FL140, 2023-05-18:
I spent a whole day selecting a proper e-Fuse IC there are a lot of those but you need to dig deep in the datasheets to find something that is not too expensive and has a small footprint. For my solution I will go with a TPS25947x variant it has a good price and is very small (still not sure if I should go with the latching or non latching variant though). (Note: I will run the feeders on a 12V rail, this chip is good up to 23V). I also looked for a eFuse with reverse polarity protection feature inlcuded (AKA 2 MOSFETS).

For a 24V Rail I found the TPS26600RHF but it is not available and large. If you don't care for the RPP then there should be other solutions.

The more I look into the power supply the more I would highly suggest to switch to 12V for the power rail. If the power rail is feeded in the middle (25 feeders to one side/25 feeders on the other side the amps on on the rails is exactly the same as for 24V). BUT all the components get much smaller, cheaper more efficient and the DC/DC converter can be skipped all together (just requires switching out the motor driver for a 12V solution, but that is a no-brainer using a DRV8220. I will write on this more tomorrow in a separate issue.

30350n, 2023-05-18:
> In regards to reducing the capacitance on either side of the buck converter, I could see trying to reduce them, but I'm hesitant to stray from the suggested values in the datasheet.

The values from the datasheet aren't suggestions though, but merely an example. The actual values for your specific application should be obtained by going through the formulae in [9.2.1.2 Detailed Design Procedure](https://www.ti.com/lit/gpn/mc34063a) (page 11).

I didn't actually do that though, but simply used an [existing online calculator](http://www.nomad.ee/micros/mc34063a/index.shtml) for the IC, to estimate the effect of switching out the capacitor:
[image]
According to this one, using 100uF of output capacity will result in ~50mVpp of ripple at 750mA, instead of ~10mVpp with 470uF.