opulo-inc/feeder #12: Simplify BOM, One 24V to 10V Buck/Boost Converter, and Atmega328

CurlyTaleGames, 2023-05-10:
# Version Number
1.01

# Bugfix or Enhancement
Enhancement

# Description & Suggested Solution

The biggest issue with designing feeders is you need a lot of them, so anything you can do to simplify the BOM goes a long way. These changes I'm suggesting are pretty big, and I don't expect you to do them, but they are worth considering.

It is excessive to have individual 24v to 10v converters for each feeder. You aren't using 24v on the feeders so it doesn't make sense to add all those extra components. You aren't using much power on the feeders, and you typically are only using 1 maybe 2 feeders at a time, so you should be able to have 1 buck/boost converter power all of them. I don't know the power usage of a single feeder, but the 1.5 amps from the MC34063AD might be a little too small to power 50 feeders.

I'm sure you've thought of it, but I would look at switching the STM32F031C6T6 to an ATMega328.

You can use the internal 8Mhz oscillator. The ATMega328 doesn't require as many caps, is cheaper, and is easier for people to program.
[image]

unpaid-bill, 2023-05-10:
The ATmega328 is now Not Recommended For New Designs [NRFND], so would make a poor substitute.
Having said that, there are other pin compatible STM32 micros that are cheaper, readily accessible, and have a lower current consumption - which allows for potentially cheaper power circuitry. One example is the STM32G031

30350n, 2023-05-10:
> The ATmega328 is now Not Recommended For New Designs [NRFND], so would make a poor substitute.

There's the ATmega328PB though, which is still in production, cheaper and also has some more features the normal ATmega328P lacks, like [a UUID](https://onlinedocs.microchip.com/pr/GUID-CBDC1838-0100-4F26-A45A-134958193C3B-en-US-4/index.html).

30350n, 2023-05-10:
> It is excessive to have individual 24v to 10v converters for each feeder. You aren't using 24v on the feeders so it doesn't make sense to add all those extra components. You aren't using much power [...]

Regarding the buck converters, while switching the official feeders to a single converter/bus design is probably not going to happen, one way to make this kinda conversion more accessible for diyers would be to add an option to bypass the buck converter completely.

Here's an example of this from my [cost optimized fork](https://github.com/30350n/low-cost-feeder) of this project:
[image]

sphawes, 2023-05-10:
Thanks for the thoughts @CurlyTaleGames!! I hear ya on all points. 

I definitely see the appeal of an ATMEGA328, especially because of its universal support. The STM32 does have a few extra parts, but that also buys us a 32 bit processor, having the same dev platform as the motherboard, and a proper SWD debug port (it seems there are some ways to do debugging on AVR but it's a bit more involved). It can also be programmed quite easily with a USB to UART dongle, so might even be easier to program than through the ICSP.

Not only from a component count but also an EMI standpoint, I totally agree with your points about the buck converter. I've already written my thoughts about it in the [Design Decisions](https://github.com/opulo-inc/feeder/blob/main/DESIGN_DECISIONS.md#why-24-volts-bus-voltage) page, but we're also constrained to the port pinout and current limits for different voltage rails on the motherboard of which there are hundreds in the world, so we're pretty tied to that.

I'm going to close this issue as these are things we can't easily implement at this point. Thank you for taking the time to write up your thoughts!

closed 2023-05-10 by sphawes