opulo-inc/feeder #13: Condense/Remove Caps

CurlyTaleGames, 2023-05-10:
# Version Number
1.01

# Bugfix or Enhancement
enhancement

# Description & Suggested Solution

Honestly what you have is fine, but these are just a couple suggestions.

There are a lot decoupling capacitors, you might be able to remove a couple. 

You might be able to condense the 4.7uF and 10uF caps and just use one or the other which would remove 1 part from your BOM.
[image]

Magpie-81, 2023-05-10:
I strongly disagree with reducing the number of decoupling caps.
Decoupling caps are not just decoration or nice to have, they are essential for a proper operation with EMI/EMC in mind. And from personal experience, there are NOT ENOUGH decoupling caps here, especially since this is only a 2-layer board. And just glancing over the layout for now, they could be placed and attached better to actually help hardening the feeders against interference. But this will be adressed in another issue.

In fact, a very essential IC (U6) that will receive and transmit a lot of EMI from the RS485-bus has no dedicated decoupling cap and really really needs one directly at the IC.

unpaid-bill, 2023-05-10:
The decoupling on the 10V rail for both motor drivers is also seriously lacking. I agree that there should be a lot more, not less, decoupling caps.

Consolidation of capacitor values is a good idea though. It's usually just as cheap (or even cheaper!) to buy a thousand caps than it is buying 100pcs, so having a _lot_ of capacitors in the same value is essentially free.

warasilapm, 2023-05-10:
I concur with @Magpie-81 and @unpaid-bill when it comes to these decoupling capacitors. The cost optimization from removing the capacitors is small compared to the much larger potential RMA and brand image costs that would result from a PCB with EMI/EMC issues. Adding decoupling capacitors is an easy win.

sphawes, 2023-05-10:
I'll rename this issue to be about _adding_ decoupling. Seems like there are three realms of improvement:

- Adding more decoupling to the 10v rail at the motor drivers
- Adding dedicated cap to the RS-485 transceiver
- Improving cap placement around the MCU

Any other details or items to tweak?

warasilapm, 2023-05-10:
To at least partially elaborate on improving the capacitor layout...

Ideally VDDA should be filtered with a FB between the 4.7 uF and 0.1 uF capacitors for the VDDA pin. The current layout makes this difficult because of the proximity to the oscillator. However, it doesn't look like you're using any of the analogue features like the ADC? So this supply may not matter much at all.

I'd like to hear more about what decoupling is desired at the motor drivers. They have a 0.1 uF across them already, but maybe people feel like there needs to be more bulk capacitance there?

Magpie-81, 2023-05-10:
With DC motors, it's always a good idea to have a capacitor directly at the motor terminals as well, to limit the EMI of the brushes. I don't think the motors used here come with that capacitor pre soldered to the motor terminals? At least i can't see one in the pictures from the OHAI documentation. Maybe there is one on the encoder pcb for the feeding motor, but not on the peeler motor.
Probably quite a lot of work to add these to every motor, but would help a lot to eliminate noise from the brushes at the source.

FL140, 2023-05-18:
I added 10nF caps at the terminals for motors in my design. Just keep in mind that this is the second best solution. Those caps should ideally be placed directly at the motor to keep the motor wires acting as antennas. ChiHai sells the peeling motor with a PCB where the caps are at the motor. IDK if the caps are also on the encoder board though.