opulo-inc/feeder #14: Vias in pads need to be avoided

Magpie-81, 2023-05-10:
# Version Number
1.0.1

# Bugfix or Enhancement
Bugfix

# Description
There are several components where there are vias in pads.
This is a known issue to cause soldering problems, as the paste and flux will get sucked away and flows into the via during the reflow process. This can lead to bad quality soldering and voids that are sometimes hard to see or detect at all without x-ray board inspection (leadless/QFN components). 
Components where this is an issue:
U1 (QFN package)
U4
U7
U8
[image]

# Suggested Solution
Avoid vias in pads or order the expensive pcb with plugged&capped vias. For the motor drivers, i would simply move the vias out of the central pad and place them to the left and right side, so there is a solder mask bridge between the exposed pad and the vias.
For U4, there is simply no need to have vias in pads there, just remove them.
For ICs like U1, i prefer to make a special footprint that covers the "in-pad vias" with soldermaks and define the paste mask to avoid printing over these areas. This would look something like this:
[image]

sphawes, 2023-05-10:
@Magpie-81 Thank you for the note! Removing those vias shouldn't be too bad. We're pretty tight on routing space under U1, but your trick is a cool way to buy back a bit of space!

warasilapm, 2023-05-10:
Moving the vias out of the pads on U4 is definitely the way to go. If they are supposed to serve the purpose of thermal vias, I would also add more and dedicate a small plane area on the backside and more vias under the part to aid in dissipation. Choose your favorite stitching pattern and stitch away.
[image]
[image]

As far as U1 goes... I'm not even sure this is necessary. The STM32F031 doesn't dissipate much power, and the datasheet does not explicitly recommend thermal vias. I would just remove them here.

If thermal vias to the ground plane are desired, they could just be placed outside of the pad for simplicity's sake.
[image]

Usually when I do this the vias are either filled or the intersections of the paste stencil cover them, which helps with wicking problems if you're using SAC (as its mobility tends to be lower than leaded).
[image]

For U7 and U8, things get a bit more complicated. There's so little area on those pads that you probably can't get away with vias directly under them without filling them. I don't really like the approach @Magpie-81 suggested with the tented vias here. Tenting them with solder mask seems likely to lead to issues with these small DFNs not laying flat and wetting incorrectly in the paste. For these, I'd probably just place thermal vias just outside the pads. This is probably sufficient especially considering the expected stall current of the motors. The drivers have thermal shutdown protection for when there is a fault condition so its not our job to design the thermal dissipation for that.
[image]

MattStep90, 2023-12-29:
Via in pad should only be avoided if the component is not designed for them. Leadless packages with thermal pads are designed for such and as long as you follow the manufacturers guidelines keeping these vias should be alright.

Vias should not be removed from U1, U7, or U8. These are meant for better thermal dissipation and small voids in thermal pads will not make a large difference in thermal performance. Moving the via's outside of the pad will however reduce thermal performance.  Suggest to instead change them to smallest via's a standard board can use (0.25mm/0.5mm usually).

Adding soldermask in the thermal pad of a leadless component is more risky as it creates the possibility for the part to not seat properly, especially with a HASL finished board.  In addition, studies have shown that attempting solder mask defined via's actually results in higher voiding.

https://www.circuitinsight.com/pdf/Impact_Via_Pad_Design_QFN_Assembly_smta.pdf

U4 absolutely should not have vias in the pad as none of those are specifically a thermal pad.