opulo-inc/feeder #15: Use a slower RS485 transceiver

Magpie-81, 2023-05-10:
# Version Number
1.0.1

# Bugfix or Enhancement
Enhancement

# Description
The MAX3078E is overkill and to fast for this application. It can work at up to 16MBPS, but the bus is only driven at 0.06MBPS. Increasing the currently used baudrates will not really offer any benefit for this application as the traffic is predictably simple and the mechanics of the rest of the machine can't handle thousands of feed operations per second anyway. So using more than 500KBPS is not an option even in the future.
Using a fast transceiver can cause more issues with EMI and signal integrity, as the signal rise/fall times will be driven to allow the very high baudrates. So the signals have way more high frequency spectrum.

# Suggested Solution
Using a lower speed transceiver like the MAX3072E with slew rate limit would send out less unnecessary high frequency signals making the bus and communication more robust and reduce the risk of EMI/EMC issues.

sphawes, 2023-05-10:
Hey @Magpie-81! Thanks for the note! That's a super good point, it is pretty overkill for the speeds we're running. I've got a few thoughts on this:
1. Feeders are the main thing we're looking to communicate with over RS-485, but having the option to communicate with other devices/peripherals at higher speeds would be a great option to have. I don't think this is likely though; a peripheral that requires communication faster than what we're already sending to feeders would likely be an entirely separate system.
2. It seems like the MAX3078E is actually _cheaper_ than its slower counterpart! LCSC has them cheaper by about 3 bucks, and Digikey has them cheaper by a bit more than a dollar (both at QTY 1 price breaks).
3. Had not considered the EMI effects of a faster signal! It hasn't caused any issues on the machine, but for certification purposes it's likely to matter more. Hard to gauge the value of making the switch for this reason without a test, but that's something I can look into.

qwertymodo, 2023-08-16:
The [HGSEMI version of the MAX3078E](https://www.lcsc.com/product-detail/_HGSEMI-_C336020.html) fulfills both being slower (_by 10x!_) and cheaper than the Analog Devices part.