opulo-inc/feeder #16: Missing capacitor on the reset signal (EMI hardening)

Magpie-81, 2023-05-10:
# Version Number
1.0.1

# Bugfix or Enhancement
Bugfix

# Description
Reset-Pin of the MCU (NRST, Pin 7) has no capacitor.
This is a common failure point with ESD and other EMI-events that will trigger an unexpected hardware reset of the controller easily. Adding capacitors help to keep the signal stable so fast transients can not trigger the edge detection interrupt that resets the controller.

# Suggested Solution
Add a 470pF and 100nF capacitor in parallel and as close as possible to the NRST Pin of the MCU.

sphawes, 2023-05-10:
Thanks for the note @Magpie-81! Added to the queue for the next rev.

warasilapm, 2023-05-10:
This is a good find. However, the 470 pF and external pullup are both unnecessary as the STM32F031 has an internal pull up and filtering on the input. I would just swap out the 10 kΩ pullup to 3.3 V to a 0.1 uF capacitor to ground as recommended by ST.
[image]

Magpie-81, 2023-05-10:
I know from experience in the EMC lab that having only a 100nF on the reset pin does not work with STM32 controllers. Having footprints and options to add more external filtering is always preferred to having to respin a board because it failed certification. The passive components don't make the BOM significantly bigger or more expensive (the 470pF is already used in the schematic, thats why i went with this value here), and space does not seem that thight here that you need to start cutting parts.

warasilapm, 2023-05-10:
We have conflicting anecdotal experience. We have a board where this reset capacitor was omitted and it made it through some pretty aggressive testing without any issue. Since nothing was connected to that pin the antenna length was much shorter than here with a reset switch. Like all EMC things there's a certain amount of black magic involved at my level of understanding and tools. This is probably also capacitor property dependent.

If the 470 pF is already in the BOM that's an easier sell IMO. If X/Y caps are added to the RS-485 lines as in #19 it could be good for them to be the same value as these.

commit b06d74ba (2023-05-17): fixes #16. adds cap to reset line for EMI hardening