opulo-inc/feeder #18: Add pull-down resistor to the RS485 driver enable signal

Magpie-81, 2023-05-10:
# Version Number
1.0.1

# Bugfix or Enhancement
Bugfix

# Description
Feeders can be hot plugged into a live system. By adding a pulldown to the driver enable signal of the RS485 driver, the chance that the driver joining the bus might send a glitch pulse on the bus is reduced. 
The IO-Pin of the microcontroller will be floating, until its firmware and clock has booted up and it configured it's IO-Pins, and that time might be long enough for the DE signal to float into an active state.

# Suggested Solution
Put a 10K pulldown on the DE signal.
The motor driver enable already has a pulldown like that in place.

warasilapm, 2023-05-10:
This is unnecessary as the MAX3078E used has hot swap circuitry to support this. See page 15 [here](https://www.analog.com/media/en/technical-documentation/data-sheets/max3070e-max3079e.pdf).

Magpie-81, 2023-05-10:
I did not check the datasheet in detail, so thanks for that feedback.
I would however still put a footprint for a resistor there. The MAX3078E uses a pretty standard footprint and pin layout, so it could get replaced with another transceiver at some point. Maybe that replacement part would benefit from already having the resistor footprint ready and connected.

warasilapm, 2023-05-10:
It seems like a no-brainer to add a pad to me. CYA pads like this are free and a good thing to get into the habit of adding to a design.

sphawes, 2023-05-12:
thanks for the thoughts @Magpie-81! I'll see if we have some board real estate to add this. things are quite tight but i should be able to squeeze this in!