opulo-inc/feeder #19: Add options for X- and Y- capacitors to RS485 data lines

Magpie-81, 2023-05-10:
# Version Number
1.0.1

# Bugfix or Enhancement
Enhancement

# Description
For EMC hardening, you usually need X- and Y-capacitors on the bus lines.
For most applications with longer transmission lines, and for RF emission and immunity, you would usually add a common mode inductor for every transceiver as well, but this might be optional in this application here. 
A common industrial RS485 interface looks more like this example:
[image]
TVS directly at the connection/terminal. Then some series element, here drawn as PTC, but we usually use 10R pulse proof resistors in our designs. (pulse proof because we apply up to 2kV surge pulses to our bus in EMC testing for compliance.)
Then there is the common mode inductor, we usually place a 2x 4.7mH inductor here (WE-SL2 series from Würth as an example).
And last but not least, the X (between the differential pair) and Y (signal to GND) capacitors. These can be anywhere between 100pF and 1nF, depends a bit on the application, but using 3 100pF would be a good starting point.

# Suggested Solution
At least add the footprints to be able to add the X- and Y- capacitors. If you don't want to populate them, that is okay, but having a footprint in place does not cost a cent, just some board space.
The common mode choke would be nice to put in there as well, but requires you to place parallel 0 Ohm resistors if you don't want to populate the inductor.

warasilapm, 2023-05-10:
Looks like the 20 Ω resistors `R15` and `R16` are filling in for the PTC in the existing designs. The common mode choke is cost prohibitive and probably unnecessary in this application, I agree.

sphawes, 2023-05-12:
Thanks for the note @Magpie-81! I believe we have some space around the transceiver to add these in. If we have space for the footprint, we'll absolutely populate them.

FL140, 2023-05-19:
@Magpie-81 since the the bus termination will be placed on slot 50 and not on a feeder which value for the termination resistor would you suggest? The feeder acts as a stub on the bus so do the 2x10 Ohm matter or not? THX

Magpie-81, 2023-05-19:
> @Magpie-81 since the the bus termination will be placed on slot 50 and not on a feeder which value for the termination resistor would you suggest? The feeder acts as a stub on the bus so do the 2x10 Ohm matter or not? THX

You just want to get the bus impedance to around 120 Ohms in total. So if you place the series resistors to protect the IC, and put the termination directly at the IC-pins, you just reduce the termination resistor value. If you have 2x 10 Ohm in series, the termination resistor would then get reduced to a 100 Ohm.
Of course, if you really want to do it properly, you would have to do some measurement and fine tuning, but at the speeds and bus length that can be expected on the feeders, it's probably not as critical as when you run a few hundred feets of cable across a building.

commit af06af9a (2023-05-17): fixes #19. adds X and Y caps on RS485 bus.