opulo-inc/feeder #26: Remove copper pour under spring connector on main board.

qwertymodo, 2023-05-18:
## Version Number
1.0.0

## Bugfix or Enhancement
Safety improvement

## Description
It is possible for the front edge of the main feeder board to rub against the edge of the feeder floor, which can expose the copper layer.  This has the potential to lead to shorts and other issues.
[image]

## Suggested Solution
There is no need for copper under the spring connector, it should be easy to just carve out the ground pour for that section of the board, on both sides.

warasilapm, 2023-05-22:
Just as a follow up on the spring connector - is the alignment and strength provided by this SMD part sufficient, or would a part with either alignment holes/through holes provide more consistent spring connector connections?

sphawes, 2023-05-24:
Awesome, thank you for the note @qwertymodo! Good point, no need for copper there, removing it reduces the chances of wearing down to metal. The bump is there to protect the spring pins and it's the exact right size to ensure they aren't hit by the extrusion when mounting.

The connector has been very consistent in our insertion tests and such. The frame print and extrusion set a hard stop limit so it's ideally impossible to have the pins depressed too far. If they have a variant of this connector with alignment pins, I'd be all for it, but wasn't able to find something like that.

Change has been made in the `rev12-mobo` branch!:
[image]