opulo-inc/feeder #30: Redesign so standard M3x14 can be used instead of non standard M3x15.

FL140, 2023-05-20:
## Version Number
Feeder Rev.12 Branch 20.05.2023 ~01:00 UTC

## Bugfix or Enhancement
Enhancement

## Description
Currently M3x15 are used for the Feeder. This is resulting in two problems.
1. The M3x15 is a length not defined in the ISO 10642 / DIN7991 norm. As a result those are hard to source.
2. The length of the screws has a +/- tolerance. This is a problem if the 15mm overall width should be honored as the tolerance is up to 0.35mm which in the worst case (low quality charge) would lead to +8,75mm with 25 feeders (one side).

## Suggested Solution
Redesign the feeder floor in a way so that standard length screws can be used and the tolerances can be honored. (There was a discussion about using heat inserts instead of the hex nuts. This would be a clean solution, but IDK if this option would be more expensive or would require more work during assembly. On the other hand the heat inserts are longer so the shorter screws would work fine even with flush inserts.)

sphawes, 2023-05-24:
Hey @FL140, thanks for the note! Totally hear ya on the non-standard lengths. I think the easiest solution might be to just have the hex nuts sit a bit deeper in the print so shorter 14mm screws will reach.

We've also been exploring the option of using self-threading plastic screws for ease of assembly, but that's gated behind testing how the threads hold up after many cycles to not discourage hacking. Either way, we'll look into how best to use a more standard length screw.

lucian151, 2025-01-10:
Fixed by switching to self-tapping screws

closed 2025-01-10 by lucian151